flathub/com.st.STM32CubeIDE #2: No ST-LINK detected error when trying to debug

mikemorrell2, 2020-09-14:
I installed STM32CubeIDE v1.4.0 from flathub on Fedora Silverblue 32 (32.20200914.0). I connected ST NUCLEO F030R8 board to USB and I try to run debugger on my project. I get "ST-LINK firmware verification" dialog box popup with message "No ST-LINK detected! Please connect ST-LINK and restart debug session." This board works okay on Windows 10 and Ubuntu 20.04 with deb version of STM32CubeIDE so I know it is not a hardware issue.

FakeShemp, 2020-09-14:
I think you at least have to plug any devices in before starting the app. Hotplugging is not supported.

mikemorrell2, 2020-09-15:
> I think you at least have to plug any devices in before starting the app. Hotplugging is not supported.

Yes, I did plug it in before launching STM32CubeIDE app. This does not appear to be the issue.

FakeShemp, 2020-09-17:
It'd be nice if you could run `flatpak run com.st.STM32CubeIDE` and post the logs that spit out. Thank you for testing :)

mikemorrell2, 2020-09-17:
I don't get any logs on the command line when I run it or when I start debugging. I do get some eclipse logs when I start-up STM32CubeIDE. There is an error log for each of the ttyXX devices, I posted one here as an example. I am using the devault debug run configuration that is set to launch a local GDB server. Not sure if the server is getting started or not. What more can I do to help test this? Is it working for anyone else currently? Maybe it is a permission issue, SELinux config issue, or is it a missing file or path config issue. Not sure what steps to try to test further. Do I need to add some options to the flatpak command line to get more logs?

eclipse.buildId=Version 1.4.0
java.version=1.8.0_252
java.vendor=AdoptOpenJDK
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_US
Command-line arguments:  -os linux -ws gtk -arch x86_64

com.st.stm32cube.ide.mpu.remote.serial
Error
Thu Sep 17 11:50:30 EDT 2020
Failed to parse udevadm output for /dev/ttyACM0

java.io.IOException: Cannot run program "udevadm": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at com.st.stm32cube.ide.mpu.remote.serial.internal.SerialPortProviderLinux$PortInfo.hasValidManufacturer(SerialPortProviderLinux.java:84)
	at com.st.stm32cube.ide.mpu.remote.serial.internal.SerialPortProviderLinux$PortInfo.validate(SerialPortProviderLinux.java:68)
	at com.st.stm32cube.ide.mpu.remote.serial.internal.SerialPortProviderLinux.run(SerialPortProviderLinux.java:156)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 5 more

mikemorrell2, 2020-09-17:
I ran flatpak with verbose flag. Here is the start-up logs. There were no additional logs when I launch debug.

[mike@mustang ~]$ flatpak run -v com.st.STM32CubeIDE
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /var/home/mike/.local/share/flatpak
F: Opening user flatpak installation at path /var/home/mike/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /var/home/mike/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Cleaning up unused container id 3984786414
F: Allocated instance id 4247325220
F: Add defaults in dir /com/st/STM32CubeIDE/
F: Add locks in dir /com/st/STM32CubeIDE/
F: Allowing host-fs access
F: Allowing homedir access
F: Allowing x11 access
F: Allowing pulseaudio access
F: Pulseaudio user configuration file '/var/home/mike/.config/pulse/client.conf': Error opening file /var/home/mike/.config/pulse/client.conf: No such file or directory
F: Allowing session-dbus access
F: Running 'bwrap --args 38 xdg-dbus-proxy --args=40'
F: Running 'bwrap --args 38 stm32cubeide'

erazemk, 2020-11-01:
+1 with this problem, plugged the board before starting the flatpak, here's the output:

```sh
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/erazemk/.local/share/flatpak
F: Opening user flatpak installation at path /home/erazemk/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/erazemk/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Cleaning up unused container id 1158913066
F: Allocated instance id 2363508423
F: Add defaults in dir /com/st/STM32CubeIDE/
F: Add locks in dir /com/st/STM32CubeIDE/
F: Allowing host-fs access
F: Allowing x11 access
F: Allowing pulseaudio access
F: Pulseaudio user configuration file '/home/erazemk/.config/pulse/client.conf': Error opening file /home/erazemk/.config/pulse/client.conf: No such file or directory
F: Allowing session-dbus access
F: bwrap --args 36 = --ro-bind /boot /boot --ro-bind /dev /dev --ro-bind /home /home --ro-bind /proc /proc --bind /run /run --ro-bind /sys /sys --ro-bind /root /root --symlink usr/bin /bin --symlink usr/lib /lib --symlink usr/lib64 /lib64 --symlink usr/sbin /sbin --ro-bind /etc /etc --ro-bind '/lost+found' '/lost+found' --ro-bind /media /media --ro-bind /mnt /mnt --ro-bind /opt /opt --ro-bind /srv /srv --bind /tmp /tmp --ro-bind /usr /usr --bind /var /var --bind /run/user/1000/.dbus-proxy/ /run/user/1000/.dbus-proxy/ --file 34 /.flatpak-info
F: bwrap --args 38 = --fd=37 unix:abstract=0004f /run/user/1000/.dbus-proxy/a11y-bus-proxy-QL5TS0 --filter --sloppy-names --call=org.a11y.atspi.Registry=org.a11y.atspi.Socket.Embed@/org/a11y/atspi/accessible/root --call=org.a11y.atspi.Registry=org.a11y.atspi.Socket.Unembed@/org/a11y/atspi/accessible/root --call=org.a11y.atspi.Registry=org.a11y.atspi.Registry.GetRegisteredEvents@/org/a11y/atspi/registry --call=org.a11y.atspi.Registry=org.a11y.atspi.DeviceEventController.GetKeystrokeListeners@/org/a11y/atspi/registry/deviceeventcontroller --call=org.a11y.atspi.Registry=org.a11y.atspi.DeviceEventController.GetDeviceEventListeners@/org/a11y/atspi/registry/deviceeventcontroller --call=org.a11y.atspi.Registry=org.a11y.atspi.DeviceEventController.NotifyListenersSync@/org/a11y/atspi/registry/deviceeventcontroller --call=org.a11y.atspi.Registry=org.a11y.atspi.DeviceEventController.NotifyListenersAsync@/org/a11y/atspi/registry/deviceeventcontroller
F: Running 'bwrap --args 36 xdg-dbus-proxy --args=38'
F: bwrap --args 36 = --ro-bind /var/lib/flatpak/runtime/org.gnome.Platform/x86_64/3.38/670bab06de70a849266b2e81f78dbf6fb68ecc55fc91bfcad062f4bf168daa53/files /usr --lock-file /usr/.ref --ro-bind /var/lib/flatpak/app/com.st.STM32CubeIDE/x86_64/stable/52bbd14a96597dafa2fe62b99ba0d30dcf4138e87ffda0992f688460bfe0a255/files /app --lock-file /app/.ref --tmpfs /usr/lib/x86_64-linux-gnu/GL --ro-bind /var/lib/flatpak/runtime/org.freedesktop.Platform.GL.default/x86_64/20.08/13b663698f585858f14e30f70bbd07b07e3d4825617115e803acd4d4a2ed5795/files /usr/lib/x86_64-linux-gnu/GL/default --ro-bind /var/lib/flatpak/runtime/org.freedesktop.Platform.VAAPI.Intel/x86_64/20.08/e21a5a785d1f04068a57705cf2444cf92178d27e1c08a543c6fbeaf8d2800eb2/files /usr/lib/x86_64-linux-gnu/dri/intel-vaapi-driver --ro-bind /var/lib/flatpak/runtime/org.freedesktop.Platform.openh264/x86_64/2.0/73f998362a6fc0d57e0c7e83e928d32b0ec14d10d0d94291033976bdcecc6b6b/files /usr/lib/x86_64-linux-gnu/openh264 --ro-bind /var/lib/flatpak/runtime/org.gnome.Platform.Locale/x86_64/3.38/cfee66614ff3d7fea5a4df1f817e8fdad4190e68ade3325eb72da26e5f263e3f-en-sl/files /usr/share/runtime/locale --tmpfs /usr/share/runtime/share/themes --ro-bind /var/lib/flatpak/runtime/org.gtk.Gtk3theme.Adwaita-dark/x86_64/3.22/7c055bbc0667524f34ca715abd67473480de93a176797a8fe4b7f7a71c4e3324/files /usr/share/runtime/share/themes/Adwaita-dark/gtk-3.0 --ro-bind-data 15 /run/flatpak/ld.so.conf.d/runtime-001-org.freedesktop.Platform.GL.default.conf --symlink /usr/lib/x86_64-linux-gnu/GL/default/vulkan/icd.d/intel_icd.x86_64.json /usr/lib/x86_64-linux-gnu/GL/vulkan/icd.d/intel_icd.x86_64.json --symlink /usr/lib/x86_64-linux-gnu/GL/default/vulkan/icd.d/radeon_icd.x86_64.json /usr/lib/x86_64-linux-gnu/GL/vulkan/icd.d/radeon_icd.x86_64.json --symlink /usr/lib/x86_64-linux-gnu/GL/default/glvnd/egl_vendor.d/50_mesa.json /usr/lib/x86_64-linux-gnu/GL/glvnd/egl_vendor.d/50_mesa.json --symlink /usr/lib/x86_64-linux-gnu/GL/default/OpenCL/vendors/mesa.icd /usr/lib/x86_64-linux-gnu/GL/OpenCL/vendors/mesa.icd --symlink /usr/lib/x86_64-linux-gnu/GL/default/lib/dri/i915_dri.so /usr/lib/x86_64-linux-gnu/GL/lib/dri/i915_dri.so --symlink /usr/lib/x86_64-linux-gnu/GL/default/lib/dri/i965_dri.so /usr/lib/x86_64-linux-gnu/GL/lib/dri/i965_dri.so --symlink /usr/lib/x86_64-linux-gnu/GL/default/lib/dri/iris_dri.so /usr/lib/x86_64-linux-gnu/GL/lib/dri/iris_dri.so --symlink /usr/lib/x86_64-linux-gnu/GL/default/lib/dri/kms_swrast_dri.so /usr/lib/x86_64-linux-gnu/GL/lib/dri/kms_swrast_dri.so --symlink /usr/lib/x86_64-linux-gnu/GL/default/lib/dri/libgallium_dri.so /usr/lib/x86_64-linux-gnu/GL/lib/dri/libgallium_dri.so --symlink /usr/lib/x86_64-linux-gnu/GL/default/lib/dri/libgallium_drv_video.so /usr/lib/x86_64-linux-gnu/GL/lib/dri/libgallium_drv_video.so --symlink /usr/lib/x86_64-linux-gnu/GL/default/lib/dri/libmesa_dri_drivers.so /usr/lib/x86_64-linux-gnu/GL/lib/dr [... cut ...] /systemd/journal/stdout --ro-bind /usr/share/fonts /run/host/fonts --ro-bind /usr/lib/fontconfig/cache /run/host/fonts-cache --tmpfs /run/host/user-fonts-cache --remount-ro /run/host/user-fonts-cache --ro-bind-data 34 /run/host/font-dirs.xml --ro-bind /usr/share/icons /run/host/share/icons --ro-bind /home/erazemk/.local/share/icons /run/host/user-share/icons --symlink /app/lib/debug/source /run/build --symlink /usr/lib/debug/source /run/build-runtime
F: Running 'bwrap --args 36 stm32cubeide'
```

alpemwarrior, 2021-03-14:
I believe you can't run udevadm inside a flatpak and thus it fails to find a programmer. Why it is trying to parse every device node in /dev is beyond me.

Maldus512, 2021-06-30:
I'm having the same problem. Does this mean that it's simply not possible to flash a device from the flatpak application?

PR #18 (2023-01-14): fix usb flashing and various updates
This should fix #2 #15 and #16 
In order for usb flashing to work, the user needs to manually add a udev rule specific to the used device, for which I added a brief explanation to the application description.
Things that were changed:

- add 'libusb', which seems to be required by st-link (st-link server won't start without it)
- update org.gnome.Platform runtime to version 42
- update STM32CubeIDE to 1.10.1
- update java to version 17
- add `unset FLATPAK_SANDBOX_DIR` to the launch script. This works around "flatpak detection" added to eclipse [here](https [... cut]
closed 2023-01-14 by FakeShemp