flathub/com.st.STM32CubeIDE #9: "Can't find bus: Cannot autolaunch D-Bus without X11 $DISPLAY" error since v1.7.0

julian-poidevin, 2021-09-06:
Since the update of the package to v1.7.0, I'm having the following error when trying to run `com.st.STM32CubeIDE` 👍🏻 

```
/home # flatpak run --verbose com.st.STM32CubeIDE --launcher.suppressErrors -nosplash -application org.eclipse.cdt.manag
edbuilder.core.headlessbuild -data '/home/tmp/' -import '/home/user/' -build DSSC-USER/Debug
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /root/.local/share/flatpak
F: Opening user flatpak installation at path /root/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /root/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Regenerating ld.so.cache /root/.var/app/com.st.STM32CubeIDE/.ld.so/fe13557fd39a2868900bb8da1ee1928e94ffbd1382f4a89da25b83eff90d0d04
F: Running: 'bwrap --args 17 ldconfig -X -C /run/ld-so-cache-dir/fe13557fd39a2868900bb8da1ee1928e94ffbd1382f4a89da25b83eff90d0d04.hFiznc'
F: Allocated instance id 3147075564
F: Add defaults in dir /com/st/STM32CubeIDE/
F: Add locks in dir /com/st/STM32CubeIDE/
F: Allowing host-fs access
F: Allowing x11 access
F: Allowing pulseaudio access
F: Pulseaudio user configuration file '/root/.config/pulse/client.conf': Error opening file /root/.config/pulse/client.conf: No such file or directory
F: Pulseaudio user configuration file '/etc/pulse/client.conf': Error opening file /etc/pulse/client.conf: No such file or directory
F: Could not find pulseaudio socket
F: Failed to run in transient scope: No systemd user session available, cgroups not available
F: Running 'bwrap --args 31 stm32cubeide --launcher.suppressErrors -nosplash -application org.eclipse.cdt.managedbuilder.core.headlessbuild -data /home/tmp/ -import /home/user/ -build DSSC-USER/Debug'
```

I'm running flatpak on a docker image built on `alpine:latest`.
The docker is running in privileged mode using this command line : `docker run -t -i -d --privileged stm32cubeidecidocker`.

I can't tell if this is this related to the new package version, I'll try with the older version and get back with more info

julian-poidevin, 2021-09-06:
This is clearly related to the latest version, I fetch the package info using `flatpak remote-info flathub --log com.st.STM32CubeIDE` :

```
flatpak remote-info flathub --log com.st.STM32CubeIDE
        ID: com.st.STM32CubeIDE
       Ref: app/com.st.STM32CubeIDE/x86_64/stable
      Arch: x86_64
    Branch: x86_64
Collection: org.flathub.Stable
  Download: 1.4 GB
 Installed: 3.2 GB
   Runtime: org.gnome.Platform/x86_64/40
       Sdk: org.gnome.Sdk/x86_64/40

    Commit: 4e989d86140354140a6c88f547a909345439d63ba4de367f2c172bf93647c85e
    Parent: eb17167edc6bc00db40430f40157e67cf02f26bbbc2feb6eef644414bb7dec7f
   Subject: remove fedc (af373b9d)
      Date: 2021-09-05 17:26:31 +0000
   History:

    Commit: eb17167edc6bc00db40430f40157e67cf02f26bbbc2feb6eef644414bb7dec7f
   Subject: Update gnome to 40 (a8c70c3a)
      Date: 2021-03-26 19:37:15 +0000

    Commit: 9986df01a78d08217616e00ed5ead80aabe93543730ea194cbe0e8a1b3684fc8
   Subject: Add stlink-server (887c26bd)
      Date: 2021-03-17 14:17:50 +0000

    Commit: 52bbd14a96597dafa2fe62b99ba0d30dcf4138e87ffda0992f688460bfe0a255
   Subject: Update runtime to 3.38 (3a83d89a)
      Date: 2020-09-17 19:25:27 +0000

    Commit: ea338abe90357740193197d306ebe41618b71816b3bd0083d7b2d3d581ab7a62
   Subject: Update com.st.STM32CubeIDE.metainfo.xml (cd539828)
      Date: 2020-07-27 15:47:23 +0000

    Commit: 793abbede06f58d4a576aefb1a166675c1be402f2191c7db1b6fba2bed4964ee
   Subject: Change image (060faafb)
      Date: 2020-05-04 13:14:06 +0000

```
I updated to the previous stable commit (eb17167edc6bc00db40430f40157e67cf02f26bbbc2feb6eef644414bb7dec7f) using :
`flatpak update --commit=eb17167edc6bc00db40430f40157e67cf02f26bbbc2feb6eef644414bb7dec7f com.st.STM32CubeIDE`

And the command now executes properly.

julian-poidevin, 2021-09-06:
I found 2 similar issue on flatpak repo :

- https://github.com/flatpak/flatpak/issues/4345
- https://github.com/flatpak/flatpak/issues/3847

However, running `flatpak update --assumeyes` as suggested in https://github.com/flatpak/flatpak/issues/3847#issuecomment-778621620 did not solve the issue

huanghaiyang, 2024-06-27:
How to resolve it .

closed 2024-06-28 by julian-poidevin